dgatf/msrc #154: GPIO decoding help needed

Richrfl, 2025-04-22:
I am not finding a way to decode the number received from the GPIO (example 39, if I have GPIOs 17, 19, 20, and 22 in ONE) into logical switches or Telemetry Calculated Sensors for each of the 6 GPIOs available.
Unless you have already figured out how to decode the number, it may require sending a package per GPIO, as one of the methods you suggested at the very beginning of this enhancement.

dgatf, 2025-04-22:
Ok. It'll send a packet for every switch enabled at application id 50nn, where nn is the number of gpio.

Richrfl, 2025-04-22:
> Ok. It'll send a packet for every switch enabled at application id 50nn, where nn is the number of gpio.

I think this is a good idea, as it will be much easier to create the DIY sensors for each GPIO.
Let me know when you have that done and where from I download the new FW/SW

dgatf, 2025-04-22:
Ok

dgatf, 2025-04-25:
Implemented with commit 26aad96a9bc7301b2cb47f84008ca909f9b899a0.

Now a packet is sent for every gpio enabled. Only [firmware](https://drive.google.com/file/d/1U6SIOXrxWNSM_oTFpoiJkpZdhAzVZ5Iy/view?usp=sharing) is updated. No need to download msrc_gui again.

Richrfl, 2025-04-25:
OK. I will come back to you later today.

closed 2025-04-25 by Richrfl