dgatf/msrc #157: Flycolor Dragon

Richrfl, 2025-05-01:
Any way to connect the Flycolor Dragon to the RP 2040?

dgatf, 2025-05-01:
Flycolor Dragon ESC? Do they have telemetry?

Richrfl, 2025-05-01:
Kind of. Check this out:

[image].

Unfortunately, I lost the app, and it is not available anymore, therefore I cannot tell you which parameters you can get as what they call "real-time data"

dgatf, 2025-05-01:
Do you have the flycolor wifi module?

dgatf, 2025-05-01:
You can find the apk (for android) [here](https://www.lemonfpv.com/h-nd-58.html)

dgatf, 2025-05-01:
If you don't have the wifi module you can try with a serial monitor at the wifi signal wire and grounded. Msrc has built in serial monitor at gpio 5. Select _ESC Protocol-> Serial Monitor_. Try with different baudrates.

Richrfl, 2025-05-01:
Great lead for the app. I do have the module, but I had trouble connecting when I had the app; that is why I uninstalled it. Anyway, I will try again with the link you gave me and also and more importantly, with the RP2040, as you suggested, which I like much better.
Thanks a lot for your help.

Richrfl, 2025-05-02:
Daniel, I think I have the wrong MSRC SW again. In the ESC pull down I have PWM, which I guess would be Serial Monitor. Not a big deal, but the only place where I can set the Baud rate is in the GPS sensor, and unless GPIO 5 has auto baud, I cannot set the rate for the PWM, or it is assumed to be the standard 20ms period. Can you comment, please?

dgatf, 2025-05-02:
Not sure if I understand.

You ESC has two connectors:
- The throttle wire, which contains the PWM signal from the receiver with a 20ms period. This signal is useless to decode and read the ESC telemetry.
- The wifi (telemetry) wire, which contains the telemetry values. Probably in serial protocol.

If you want a serial monitor, select _ESC Protocol -> Serial Monitor_. It will show serial monitor parameters where you can change the baudrate:

[image]

You have to connect the wifi signal to gpio 5 (and gnd).

The _ESC Protocol -> PWM_ is used to read rpms on some ESC that is transmitted through PWM in a separate wire.

Richrfl, 2025-05-02:
We are on a different page. I assumed the serial data available in the wifi signal pin, could be converted to FrSky telemetry, but it looks to me now that I can read that data into the debug screen of the MSRC SW. Is that correct?

dgatf, 2025-05-02:
We need to read the telemetry packets from your esc to decode and implement the telemetry. I need a dump of the wifi signal to do this. What you will read in the debug screen of msrc is the raw serial data. Then I need to decode it and implement for all receivers protocols.

Richrfl, 2025-05-02:
Daniel, if you want to implement this support, I will be happy to help. It seems to be a hell of a project. Please. comment.

dgatf, 2025-05-02:
Implement a esc telemetry is not a big issue. Receivers protocols are.

If you can read with serial monitor the telemetry packets, it wont be a big effort.

Richrfl, 2025-05-02:
Could this be of any use for you?

[image]

I tried from 4800 through 115200, different parities and 1 and 2 stop bits, without any luck.

dgatf, 2025-05-02:
Unfortunately not. It seems that telemetry needs to be requested by the wifi module/app. This makes thinks harder. You can try with the module attached and the app requesting telemetry. Up to you if you want to go any further.

Richrfl, 2025-05-02:
Yes. Tomorrow I will run a full list of baud rates, parity and stop bits. 
Then I will hook up a Y cable in the wifi module to look at the data exchange, and if needed, again will run the full list check out If this doesn't work, we will have to drop the ball. I wonder if there is a discrepancy in the number of data bits between the Serial monitor and the Flycolor protocol, and that is the reason I get garbled data in the debugger.

dgatf, 2025-05-02:
You get garbage because you enable the debug too soon. It isn't from the serial monitor.

Don't bother too much with parity an stop bits for now. Even if they are incorrect you should be getting some serial data. You can adjust them once you get something.

You sholud see something like:

`Serial monitor : 0x34 0xA2 ...`

Richrfl, 2025-05-02:
Well, I was lucky. Without knowing, I ran the debug at 19200 default parity and stop bit and got something:
This log, whose values are read in the app, are

voltage   V                22.8	
Current A                 0	
mos temp Celsius 28.7	
rpm		                 0	
throttle usec	        988	
PWM		         0	

USB. Debug enabled. MSRC firmware version 1.2.2
USB (84)
Serial monitor (301). Length: 1 (8576 ms):  0xAA
Serial monitor (202). Length: 1 (8603 ms):  0x53
Serial monitor (200). Length: 1 (8605 ms):  0xA
Serial monitor (200). Length: 1 (8608 ms):  0xE
Serial monitor (200). Length: 1 (8610 ms):  0x0
Serial monitor (200). Length: 1 (8613 ms):  0x0
Serial monitor (200). Length: 1 (8615 ms):  0x21
Serial monitor (200). Length: 1 (8618 ms):  0xE8
Serial monitor (200). Length: 1 (8621 ms):  0x0
Serial monitor (200). Length: 1 (8623 ms):  0x0
Serial monitor (200). Length: 1 (8626 ms):  0x3
Serial monitor (200). Length: 1 (8628 ms):  0xDB
Serial monitor (200). Length: 1 (8631 ms):  0x0
Serial monitor (200). Length: 1 (8634 ms):  0x52
Serial monitor (200). Length: 1 (8636 ms):  0x54
Serial monitor (200). Length: 1 (8777 ms):  0xAA
Serial monitor (200). Length: 1 (8805 ms):  0x53
Serial monitor (200). Length: 1 (8808 ms):  0xA
Serial monitor (200). Length: 1 (8810 ms):  0xD
Serial monitor (200). Length: 1 (8813 ms):  0x0
Serial monitor (200). Length: 1 (8816 ms):  0x0
Serial monitor (200). Length: 1 (8818 ms):  0x21
Serial monitor (200). Length: 1 (8821 ms):  0xF6
Serial monitor (200). Length: 1 (8823 ms):  0x0
Serial monitor (200). Length: 1 (8826 ms):  0x0
Serial monitor (200). Length: 1 (8828 ms):  0x3
Serial monitor (200). Length: 1 (8831 ms):  0xDC
Serial monitor (200). Length: 1 (8834 ms):  0x0
Serial monitor (200). Length: 1 (8836 ms):  0x52
Serial monitor (200). Length: 1 (8839 ms):  0x54
Serial monitor (200). Length: 1 (8979 ms):  0xAA
Serial monitor (200). Length: 1 (9008 ms):  0x53
Serial monitor (200). Length: 1 (9010 ms):  0xA
Serial monitor (200). Length: 1 (9013 ms):  0xE
Serial monitor (200). Length: 1 (9016 ms):  0x0
Serial monitor (200). Length: 1 (9018 ms):  0x0
Serial monitor (200). Length: 1 (9021 ms):  0x22
Serial monitor (200). Length: 1 (9023 ms):  0xE8
Serial monitor (200). Length: 1 (9026 ms):  0x0
Serial monitor (200). Length: 1 (9029 ms):  0x0
Serial monitor (200). Length: 1 (9031 ms):  0x3
Serial monitor (200). Length: 1 (9034 ms):  0xDD
Serial monitor (200). Length: 1 (9036 ms):  0x0
Serial monitor (200). Length: 1 (9039 ms):  0x52
Serial monitor (200). Length: 1 (9041 ms):  0x54
Serial monitor (200). Length: 1 (9184 ms):  0xAA
Serial monitor (200). Length: 1 (9211 ms):  0x53
Serial monitor (200). Length: 1 (9213 ms):  0xA
Serial monitor (200). Length: 1 (9216 ms):  0xE
Serial monitor (200). Length: 1 (9218 ms):  0x0
Serial monitor (200). Length: 1 (9221 ms):  0x3
Serial monitor (200). Length: 1 (9223 ms):  0x21
Serial monitor (200). Length: 1 (9226 ms):  0xF6
Serial monitor (200). Length: 1 (9229 ms):  0x0
Serial monitor (200). Length: 1 (9231 ms):  0x0
Serial monitor (200). Length: 1 (9234 ms):  0x3
Serial monitor (200). Length: 1 (9236 ms):  0xDC
Serial monitor (200). Length: 1 (9239 ms):  0x0
[... 118 lines cut ...]
Serial monitor (200). Length: 1 (373237 ms):  0x54

This sequence happens when I hit Connect in the app

USB. Debug enabled. MSRC firmware version 1.2.2
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
USB (84)
Serial monitor (200). Length: 20 (1492640 ms):  0xA5 0x80 0x2F 0x15 0x8 0x4 0x2 0x1 0x0 0x80 0x40 0xC8 0xC5 0x65 0x43 0x8A 0xF5 0x49 0x22 0x8B
USB (84)
Serial monitor (198). Length: 9 (1492799 ms):  0x30 0x30 0x30 0x31 0xF0 0x51 0x0 0x1 0xC4
USB (84)

I will forward more data transfer generated at the time the app is connected to the wifi ESC. I am getting some issues with the app right now.

dgatf, 2025-05-02:
Please increase timeout to 10

dgatf, 2025-05-02:
Those are telemetry packets. Keep the setup as is now. Are you using the app to get that debug? If not, no need to use it.

dgatf, 2025-05-02:
- Byte 1, header: 0xAA
- Bytes 4&5 current x10
...

dgatf, 2025-05-03:
I don't need additional info for now. I'll let you know whem implemented to test.

dgatf, 2025-05-03:
Maybe it is needed a start command. Can you get that debug without the wifi module?

dgatf, 2025-05-03:
What is your bec voltage?

Richrfl, 2025-05-03:
I will increase the TO in the next debug.
If the module is not connected, I will not get anything. Yesterday, I detected two data sequences that may be important, but I was not able to confirm them because of an issue with the app. One is when the app connects to the module, and the other is when I request the app to start receiving data.
Regardless, as you suggested, I will enable debug before powering on the ESC.
BEC voltage is 5.4.

Corrected communication setting: Baud rate: 19200   Stop bit: 1 **Parity: ODD**

dgatf, 2025-05-03:
>  BEC voltage is 5.4.

Then bytes 7 & 8 is not bec voltage. Could be consumption (mAh)? However, this is not populated in you app. I'll ignore those bytes for now.

Richrfl, 2025-05-03:
Enabling debug before power-on shows no communication.
When the app connects to the module, this is the exchange:

USB. Debug enabled. MSRC firmware version 1.2.2
USB (84)
USB (84)
Serial monitor (301). Length: 1 (537530 ms):  0xA5
Serial monitor (202). Length: 3 (537829 ms):  0x0 0x2F 0xF5
Serial monitor (202). Length: 9 (537939 ms):  0x0 0x80 0x40 0x20 0x10 0x8 0x4 0x6A 0xFF
Serial monitor (200). Length: 9 (537956 ms):  0x46 0xA6 0x56 0xA8 0x8A 0xF5 0x49 0x22 0x8B
Serial monitor (200). Length: 9 (538115 ms):  0x30 0x30 0x30 0x31 0xF0 0x51 0x0 0x1 0xC4
USB (84)
USB (84)
USB (84)
Serial monitor (200). Length: 4 (541416 ms):  0xFF 0x80 0x5E 0xE0
Serial monitor (200). Length: 2 (541431 ms):  0x12 0x8A
USB (84)
Serial monitor (200). Length: 1 (541586 ms):  0xC4
Serial monitor (200). Length: 2 (541773 ms):  0x3 0xB8
Serial monitor (200). Length: 2 (541792 ms):  0x1 0x8A
Serial monitor (200). Length: 115 (542019 ms):  0x1 0x0 0x1 0xF 0xB4 0x7 0xBC 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x55 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x1 0x4 0x0 0x0 0x1 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x0 0xD 0x23 0x50 0x4C 0x41 0x4E 0x45 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x20 0x23 0x57 0x69 0x6E 0x44 0x72 0x61 0x67 0x6F 0x6E 0x5F 0x73 0x65 0x72 0x69 0x65 0x73 0x56 0x35 0x30 0x31 0x30 0x5F 0x31 0x30 0x32 0x30 0x5F 0x30 0x34 0x20 0x20 0x20 0xF0 0xB3 0xC4
USB (84)
USB (84)

This information shows in the app screen:

Since

[image]

[image]

 I request data till I stop, I get this:

USB. Debug enabled. MSRC firmware version 1.2.2
USB (84)
USB (84)
Serial monitor (166). Length: 1 (1000463 ms):  0xAA
Serial monitor (166). Length: 14 (1000524 ms):  0x53 0xA 0xC 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDD 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1000661 ms):  0xAA
Serial monitor (166). Length: 14 (1000721 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xD7 0x0 0x0 0x3 0xDC 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1000861 ms):  0xAA
Serial monitor (166). Length: 14 (1000922 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDC 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1001061 ms):  0xAA
Serial monitor (166). Length: 14 (1001122 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDD 0x0 0x52 0x54
USB (84)
Serial monitor (166). Length: 1 (1001264 ms):  0xAA
Serial monitor (166). Length: 14 (1001324 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDD 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1001468 ms):  0xAA
Serial monitor (166). Length: 14 (1001529 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDC 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1001686 ms):  0xAA
Serial monitor (166). Length: 14 (1001747 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDC 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1001866 ms):  0xAA
Serial monitor (166). Length: 14 (1001927 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDC 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1002074 ms):  0xAA
Serial monitor (166). Length: 14 (1002135 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDD 0x0 0x52 0x54
Serial monitor (166). Length: 1 (1002267 ms):  0xAA
USB (84)
Serial monitor (166). Length: 14 (1002327 ms):  0x53 0xA 0xB 0x0 0x0 0x23 0xC8 0x0 0x0 0x3 0xDC 0x0 0x52 0x54
USB (84)
USB (84)
USB (84)

Richrfl, 2025-05-03:
Yes, it looks like each time 0xAA is issued, a complete set of parameters is sent. And that happens to be always the first line of data.
It seems that you have this one pretty much figured out.
Just to confirm, do you think you just can generate the 0xAA once to see if we get an answer from the ESC without the module?

dgatf, 2025-05-03:
>  Just to confirm, do you think you just can generate the 0xAA once to see if we get an answer from the ESC without the module?

Yes.

Which receiver are you using, frsky with smartport?

Richrfl, 2025-05-03:
Yes

Richrfl, 2025-05-03:
Something that you requested adjust the timeout to 10ms. This is the reading:

Serial monitor (192). Length: 1 (388500 ms):  0xAA
Serial monitor (192). Length: 14 (388561 ms):  0x53 0xA 0x8 0x0 0x0 0x24 0x91 0x0 0x0 0x3 0xDC 0x0 0x52 0x54

This reading is for the following values:

Voltage: 22.8
Current: 1.3
temp 26.1
RPM: 8000
Throttle: 1174
PWM  19%

dgatf, 2025-05-03:
I've added support for Flycolor in main branch. Please download updated [binaries](https://github.com/dgatf/msrc?tab=readme-ov-file#1-binaries). You need both firmware and msrc_gui. It does the same as the wifi/app module: request telemetry every 200ms.

Richrfl, 2025-05-03:
It looks like we are missing something. The 0xAA is sent, but there is no answer.

USB. Debug enabled. MSRC firmware version 1.3.0
USB (100)
Flycolor (217) > 0xAA
GPIO (208): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
USB. Processing (2) 0x30 0x3102
USB. Send config
USB (100)
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
USB (100)
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (118) > 0xAA
GPIO (102): 0x0

dgatf, 2025-05-03:
Are you sure is odd parity?

You can enable telemetry from the app, disconnect the wifi module and then connect the rp2040. This will confirm if at least the request from msrc is correct.

Richrfl, 2025-05-03:
I think what we are missing is the command that the app sends to start sending data. 

It looks like it should be a command to tell the module to start requesting telemetry data, and I guess it should be some code that will indicate if the data requested is configuration or telemetry. What do you think?

dgatf, 2025-05-03:
Yes, you should discover new sensors after selecting the esc in msrc_gui.

Richrfl, 2025-05-03:
Ok, with the module connected I get data on the telemetry screen.
Without the module as well, however we will need some trim on the parameters or timing.

Here is the log

Flycolor (98) > 0xAA
GPIO (102): 0x1
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (134) > GPIO: 17 STATE: 1 > 10 11 51 1 0 0 0 8C 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
GPIO (102): 0x1
Flycolor (98) > 0xAA
GPIO (102): 0x1
GPIO (102): 0x1
Flycolor (98) > 0xAA
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (134) > GPIO: 17 STATE: 1 > 10 11 51 1 0 0 0 8C 
GPIO (102): 0x1
USB (100)
GPIO (102): 0x1
Flycolor (98) > 0xAA
GPIO (102): 0x1
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (134) > GPIO: 17 STATE: 1 > 10 11 51 1 0 0 0 8C 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor double (108) > 10 60 B 0 0 0 0 84 
GPIO (102): 0x1
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor double (108) > 10 50 B 3C 8 0 0 50 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor (126) > 10 70 B 1A 0 0 0 5A 
Flycolor (98) > 0xAA
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor cell (122) > 10 0 3 0 2A 29 0 99 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (134) > GPIO: 17 STATE: 1 > 10 11 51 1 0 0 0 8C 
GPIO (102): 0x1
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
GPIO (102): 0x1
Flycolor (98) > 0xAA
GPIO (102): 0x1
GPIO (102): 0x1
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (134) > GPIO: 17 STATE: 1 > 10 11 51 1 0 0 0 8C 
Smartport (122) < 0x7E 0xE9 

and here is the Telemetry screen with the motor running 

[image]

dgatf, 2025-05-03:
> Without the module as well, however we will need some trim on the parameters or timing.

That means that you are getting esc telemetry after disconnecting wifi module and connecting the rp2040, isn't it? So I suppose the request is working.

But in the log I don't see the answer from the esc.

Question: with the serial monitor enabled, what happens if you disconnect the wifi after getting telemetry in the app? The serial monitor stops, isn't it?

Richrfl, 2025-05-03:
> > Without the module as well, however we will need some trim on the parameters or timing.
> 
> That means that you are getting esc telemetry in the transmitter after disconnecting the wifi module and connecting the rp2040, isn't it? So I suppose the request is working.

That is correct it is working.

> 
> But in the log I don't see the answer from the esc.

Given that there is no other sensor than C1 (GPIO 17) all we see in the log is between the 2040 and the ESC. Maybe the 0xAA is overlaping the ESC response (timing between 0xAAs?
> 
> Question: with the serial monitor enabled, what happens if you disconnect the wifi after getting telemetry in the app? The serial monitor stops, isn't it?

Correct. stops sending data.

dgatf, 2025-05-03:
> Given that there is no other sensor than C1 (GPIO 17) all we see in the log is between the 2040 and the ESC. Maybe the 0xAA is overlaping the ESC response (timing between 0xAAs?

Really odd. As long as you get live telemetry in the transmitter, is ok.

I will add the initial data sent from the app.

Question: do you get correct telemetry values? Is there any mismatch?

Richrfl, 2025-05-03:
Yes, there is a little mess on the telemetry. Take a look at the screenshot I posted above

dgatf, 2025-05-03:
Voltage and temp seems correct. Are rpms and current correct? If not, what values you expected?

Richrfl, 2025-05-03:
No, rpm and current are not right. I do not have the values, but I can give you a close number based on a new test under the same conditions:
rpm ~ 2200, current ~ 0.2 A throttle 1045, PWM ~5%

dgatf, 2025-05-03:
Wait for the github action to finish deployment...

dgatf, 2025-05-03:
I've installed the apk. There're two steps:
- 
- When you click on _Click Connect_, do you get serial telemetry?
- When you click on _Play button_ on the tyelemetry scrren

If so, please

Richrfl, 2025-05-03:
This is the log with the new fw:

USB. Debug enabled. MSRC firmware version 1.3.0
USB (100)
Flycolor (114) > 0xAA
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
GPIO (102): 0x0
Flycolor (114) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (114) > 0xAA
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor double (108) > 10 60 B 0 0 0 0 84 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor double (100) > 10 50 B 0 0 0 0 94 
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor (124) > 10 70 B 0 0 0 0 74 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor cell (122) > 10 0 3 0 0 0 0 EC 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
Smartport (122) < 0x7E 0xE9 
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Flycolor (114) > 0xAA
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (114) > 0xAA
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
Smartport (122) < 0x7E 0xE9 
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Flycolor (114) > 0xAA
GPIO (102): 0x0
USB (100)
GPIO (102): 0x0
Flycolor (114) > 0xAA
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Flycolor (114) > 0xAA
GPIO (102): 0x0
GPIO (102): 0x0
Flycolor (114) > 0xAA
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor double (108) > 10 60 B 0 0 0 0 84 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor double (100) > 10 50 B 0 0 0 0 94 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor (124) > 10 70 B 0 0 0 0 74 
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor cell (122) > 10 0 3 0 0 0 0 EC 
GPIO (102): 0x0
Smartport (122) < 0x7E 0xE9 
Smartport. Sensor GPIO (126) > GPIO: 17 STATE: 0 > 10 11 51 0 0 0 0 8D 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9

Richrfl, 2025-05-03:
> I've installed the apk. There're two steps:
> 
> * When you click on _Click Connect_, do you get serial output?

Yes

> * When you click on _Play button_ on the telemetry screen, do you get serial output?

Yes

> 
> Please post both serial outpùt (if any) for each of the two steps.

To be able to do this, either I have to use the serial monitor or fall back to the original FW. What should I do?

dgatf, 2025-05-03:
I've implemented partially the handshake. If everything goes as expected, you should see this log:
```
Flycolor (153). Handshake > 0xA5 
Flycolor (48). Handshake 0 < 0x0 0x2F 0xF5 
Flycolor (48). Handshake > 0x0 0x80 0x40 0x20 0x10 0x8 0x4 0x6A 0xFF 
Flycolor (46). Handshake 1 < 0x46 0xA6 0x56 0xA8 0x8A 0xF5 0x49 0x22 0x8B 
Flycolor (46). Handshake > 0x30 0x30 0x30 0x31 0xF0 0x51 0x0 0x1 0xC4 
Flycolor (46). Handshake 2 < 0xFF 0x80 0x5E 0xE0 
Flycolor (46). Handshake > 0x12 0x8A 
Flycolor (46). Handshake 3 < 0xC4 
```
Download updated [firmware](https://drive.google.com/file/d/1U6SIOXrxWNSM_oTFpoiJkpZdhAzVZ5Iy/view?usp=drive_link).

Richrfl, 2025-05-03:
Serial monitor or FrSky?

dgatf, 2025-05-03:
Frsky

Richrfl, 2025-05-03:
I am getting a bunch of this:

Smartport (181) < 0x7E 0xE9 
Smartport (130) < 0x7E 0xE9 
Smartport (130) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
USB (100)
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 
Smartport (122) < 0x7E 0xE9 

Nothing in the telemetry screen

dgatf, 2025-05-03:
Disconnect smartport from rp2040. 

Remember handshake happens 15 seconds after boot.

Richrfl, 2025-05-03:
I did as suggested. No telemetry connection.
The handshake does not happen (at least in 15 sec)

[image]

dgatf, 2025-05-03:
Even with nothing connected you'll see the initial handshake. See [video](https://youtu.be/bQLj-L9oXtg?si=I6e2mSnAL5eIOVfu).

Remember to reboot esc and rp2040 for every test.

Richrfl, 2025-05-03:
Yes, I can see the handshake now

dgatf, 2025-05-03:
Please post it to check it.

Richrfl, 2025-05-03:
After reseting the 2040 and powering on the ESC this is what I get (I enabled the debug before powering on the ESC):

02��%w	w�?�����������@��d*>� y$
USB. Debug enabled. MSRC firmware version 1.3.0
USB (100)
USB (100)
Flycolor (153). Handshake > 0xA5 
Cell count (254): 1
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)
USB (100)

If I power Cycle the ESC the handshake is not executed. It requires to RST the 2040.

Richrfl, 2025-05-03:
Can you generate an RST for the 2040 when it gets power from the SPort?

dgatf, 2025-05-03:
It is difficult to know who is writing in serial half duplex. Please, with the serial monitor, disconnect wifi signal module (keep vcc and gnd) and post the log. This is to see what is being sent at the beginning of the handshake from the wifi module only.

dgatf, 2025-05-03:
> Can you generate an RST for the 2040 when it gets power from the SPort?

That cant be done only with code.

Richrfl, 2025-05-03:
> > Can you generate an RST for the 2040 when it gets power from the SPort?
> 
> That can't be done only with code.

That was my concern, This is more HW. Just to keep in mind. A jumper from one of the GPIOs assigned to Contact to one of the leads of the RST button can make the trick, but let's put this in the fridge right now. I'll disconnect the Data wire from the module now.

[64 more replies]