dgatf/msrc #158: Smart esc wiring

ToniPedrosa, 2025-05-01:
Hi there,

First off, amazing work on this project—really impressive and much appreciated!

I’ve been checking out the new Spektrum Smart ESC support (actually started digging into it before switching to the main branch) and reviewing the code. Everything makes sense, except one thing: in the connection diagram, GPIO4 (TX) is shown going to the receiver, but it’s not part of the SBUS2 line (which is already on GPIO0 and 1).

So I’m wondering—where exactly should GPIO4 connect on the receiver?

I noticed in the documentation that GPIOs 4 and 5 are used for the Spektrum Smart ESC signal, which makes more sense to me than GPIO4 going to the receiver.

Since both pins are connected to the ESC signal line, I was just wondering: would it be better to add a 1kΩ resistor between them (TX and RX), as is often done in half-duplex UART setups, or is it fine to wire them directly?

Thanks again for all your work—really enjoying digging into the project.

[image]

dgatf, 2025-05-01:
I've just updated the smart esc circuit:

[image]

dgatf, 2025-05-01:
Connections:

- Connect smartport to gpio 0&1. Resistor not needed.
- Connect smart esc to gpio 4&5. Resistor not needed.
- Connect throttle signal to gpio 12.
- Connect reverse signal to gpio 13.
- Connect common ground.

The resistor (100ohm) is optional. rp2040 has internal protection resistors for gpios (~100ohm).

closed 2025-05-02 by dgatf