dgatf/msrc #159: Add Support for XDfly ESC Telemetry

garfield2412, 2025-05-09:
Hi Daniel,

Another great project to integrate the protocol in MSRC?
I'll post some serial logs these days.
These ESC seem to be more popular as time goes by (especially for Heli Pilots)
Maybee they are similar to ZTW.

dgatf, 2025-05-09:
Waiting for the logs.

garfield2412, 2025-05-09:
Some specs:

 * XDFLY
 *
 *     - Serial protocol is 115200,8N1
 *     - Big-Endian byte order
 * 
 * Frame Format
 * ――――――――――――――――――――――――――――――――――――――――――――――――――――――――
 *      0:      start byte (0xA5)
 *      1:      length in bytes (8)
 *      2:      command 
 *          0x03 handshake
 *          0x34 set_par
 *          0x33 get_par
 *      3:      param number
 *      4-5:    param value
 *      6-7:      CRC16
 * 
 * 
 * checking whether the param is valid for the connected ESC:
 *      if the highest bit of the value is set, the parameter is valid for the connected ESC
 *          bool paramValid = paramValueRaw & 0x8000 != 0 //param shall not be displayed
 *      the actual data is in the lower 15 bits
 *          uint16_t paramValue =  paramValueRaw & 0x7FFF
 * 
 * for setting a param the high bit of the value is set to 1
 *     uint16_t paramValueRaw = paramValue | 0x8000
 * 
 * when setting a param and the ESC responds with 0xFFFF, setting the param was not successful
 */

Might be a bidirectional Info because you can set parameters via the protocol too.

Curious if I have to send 0x33 and then the parameter to get a reply, or if the ESC is sending continues.

dgatf, 2025-05-09:
That is not related to telemetry, but to program the esc...

dgatf, 2025-05-09:
It seems that has few telemetry protocols to choose from.

garfield2412, 2025-05-09:
> It seems that has some telemetry protocols to choose from. It has sbus2 and jeti which are known.
> 
> Can you test the esc with sbus2 or jeti receiver?

I have an ESC, but no SBus2 hardware or Jeti.

Only JR and FrSky

dgatf, 2025-05-09:
Maybe connecting the serial monitor to Tx, as with rotorflight, it sends telemetry without being polled.

dgatf, 2025-05-16:
Were you able to read any serial output from the esc?

It seems that there is a handshake process.