dgatf/msrc #160: Throttle Signal not propagating with Spektrum Smart ESC

tstuli, 2025-05-19:
Hi.  Great project -- just trying to get it up and going on my end.

I seem to be having a problem with the throttle signal working with a Avian 60-Amp ESC.  Although the transmitters PWM is being detected by the Raspi, and seemingly being sent to the ESC, it remains in its repeating "short" beep (e.g. throttle signal not at idle).

Telemetry/etc seems to be working fine so the components appear to be talking to each other fine.

I was wondering if you've run into this or had any thoughts?

tstuli, 2025-05-19:
I've had a bit more luck replacing the 100ohms with dead shorts on the board.  I'm guessing the drive wasn't strong enough.  Now telemetry and motor control are working.  As a note I'm interconnecting a Frsky TD S6R with the Avian

One oddity however is that the motor only kicks in at pulse width around 1380us (I would have expected closer to 1000us).  I'll try and get a bit more debug on here.

Thanks for your assistance.

dgatf, 2025-05-19:
Just to be sure,  you needed to remove the 100ohm resistor from GPIO0, didn't you? I'll update the docs if so.

I'll review the pwm measurement.

dgatf, 2025-05-20:
I've removed the resistor at GPIO4 from the docs/circuit. And I've adjusted the PWM measurement based on my RP2040, reducing duration in 15us. Though this may vary from one to another. If the variation persists I can add low and high throttle duration parameters in MSRC Link to calibrate the PWM signal.

tstuli, 2025-05-20:
Thanks sounds good!  I'll try the new PWM code shortly and let you know.

Regarding the resistors, there is mention that the outputs are all protected internally on the RP2040.  Is that correct and I'm curious to see the detailed specs (was not able to find any details on the output driver design).  Otherwise, we would be relying on protection on the ESC/Receiver to prevent both outputs fighting so I'm wondering if some smaller resistance is still warranted.

Note I removed them both from GPIO4 and GPIO0 (Did not try them independently).

Thanks again for your help.

dgatf, 2025-05-20:
I've removed the PWM adjustment. Further tests shows that error in PWM measurement is less than 0.2% (2us). If the mismatch persists, I can add parameters in MSRC Link to calibrate PWM signal.

dgatf, 2025-05-20:
> Regarding the resistors, there is mention that the outputs are all protected internally on the RP2040. Is that correct and I'm curious to see the detailed specs (was not able to find any details on [...]

I measured the internal resistors adding a 100ohm resistor and measuring the current.

dgatf, 2025-05-20:
In the RP2040, the Tx is disabled except when transmitting. 90ohm should be enough to protect the GPIOs. In normal conditions both Tx are never transmitting at the same time.

closed 2025-05-21 by dgatf