dgatf/msrc #164: It's definitive not an issue but a wish to add a G-Force Sensor

Astor87, 2025-06-04:
Sorry I am pretty new using this great MSRC project.  Mainly for Vario & GPS
It is not an issue rather than a great wish and I don't know wether to place it here is correct.
Is there a possibility to add a G-Sensor may be  such as GY-521 MPU-6050 3-axis-gyro  & acceleration Sensor or similar cheap to the MSRC .
As I can see there are some pins on the RP2040 still unused. 
Would be great to have a G-Force Sensor to measure max G-Force an accelerations while flying a RC model, as in good old MPX times.
I switched from FrSky and MPX to Expresslrs so I guess Hott telemetry would be nice since CSRF doesn't supply more Sensor readings ,  if I read it right or any other good possible suggestion for ELRS
Thanks 
Wolfram

Astor87, 2025-06-05:
Wow , thanks will try !!! 
Didn‘t know that I can connect more than one Sensor to the I2C    
Great !!!!
BR
Wolfram

Astor87, 2025-06-05:
Sounds superb ! 
Many thanks in advance !
I am confident many will enjoy that !!
BR
Wolfram

dgatf, 2025-10-15:
Added support for MPU6050:
- Available attitude and acceleration in: smartport, fport, fbus, jetiex
- Available attitude in: crsf

closed 2025-10-15 by dgatf