dgatf/msrc #165: Support for more than 1 NTC sensor

Maze1982, 2025-06-05:
I fly a big gasser that have 2cylinder engine. I can se the value of having more than one ntc value, And also be able to pickup rpm value from the electronic ignition systems from like an Da or Dle engines that are popular brands. Or maybe having an separate magnetic pickup for rpm measument. maybe thats alreade possible by using the pwm option

dgatf, 2025-06-05:
Yes, you can use _ESC Protocol -> PWM_ to read rpm in PWM or pulses format. Note that I've just added support for pulses (not only PWM).

Which radio system are you using? Depending on it, it is may be possible to use two MSRC boards to get two ntc readings.

Yes, allowing to select analog inputs for each analog pin is in the TODO list. To implement this is really extensive.

Maze1982, 2025-06-05:
I am using Frsky and smartport, i didnt even think about running 2 boards.

dgatf, 2025-06-05:
Yes, you only need to select a different _Sensor Id_ in the receiver protocol parametersfor each MSRC board. Connect both to the Smartport port.

dgatf, 2025-06-05:
Electronic ignition systems from like an Da or Dle. Do they output rpms in puls or PWM or they use another protocol?

dgatf, 2025-06-06:
> I will try going with double boards and se how it works, i only have one smartport lead from the engine compartment but i think it should work anyway right

You can use servo splitter (Y connector).

Maze1982, 2025-08-29:
Some update on this, it took some time but i built a small "unity" crammed together connected with pins on the appropiate places, and a small bec in between to get the voltage down because i run 7v 
I have gotten both the temp sensors to work (it works really good). but i cant get the rpm sensor to work. I have checked the sensor and it is an hal sensor that drops its resistance to zero when passing the magnet on the flywheel. I am unsure if i need some pickup resistor on the board side? , or even were to connect it. on the pictures it seems like it should be pin 6 but in the documentation i think it points to pin 4 but really unsure here.  Best regards /Jonas

dgatf, 2025-08-29:
Yes, there was a mistake in the circuit image. PWM or pulses goes to GPIO4. Fixed with commit 02951b1.

It uses the internal pullup of the RP2040 (50k).

You should see in the log:
```
Esc PWM (98) < Rpm: 2991
Esc PWM (98) < Rpm: 2990
Esc PWM (98) < Rpm: 2990
```
If the pulses are not detected you may need to add an external pullup from 3k to 10k.

Maze1982, 2025-08-30:
No sorry i dont have any specs, its completely sealed in plastic, i "borrowed" it from an other engine.