dgatf/msrc #166: need help for the first msrc sensor

ParkerEde, 2025-06-06:
A great project. Many thanks for this.
I am trying to create my first msrc sensor RP2040Zero for frsky s.port. So far understood:
1. USB connection with BOOT button pressed
   I get a USB drive and copy the MSRC-RP2040_v1.3.uf2 to it
2. I press RESET on the RP2040Zero
3. start msrc_link_v1.3.exe
4. the COM port is recognised, I press ‘Connect’
5. the setting remains on the default FrSky Smartport, SensorID1
6. I activate the tick ‘Voltage’
7. I press the ‘Update config’ button
8. I press RESET on the RP2040Zero
9. disconnect from USB
10. I connect the RP2040Zero to the s.port of the RX according to the connection diagram
11. I search for sensors in the radio
12. only two more sensors are found in addition to the RX RSSI RxBt sensors 5113 and 5115. Both have the value 0 even if a voltage is applied via the voltage divider at PIN 26
What am I doing wrong?
Thank you for your help

ParkerEde, 2025-06-07:
I've already used 1.3.
Is the workflow I wrote right?

dgatf, 2025-06-07:
I could reproduce only once. It seems an issue with the configuration being corrupted at some byte.

Please do this. With v1.2.2, select from the _File_ menu _Restore Default Config_, then flash v1.3 and enable voltage again.

dgatf, 2025-06-07:
With commit 9d73597, I've added GPIO 15 to restore default config if is grounded at boot. This is only available from current (development) version.

ParkerEde, 2025-06-07:
> I could reproduce only once. It seems an issue with the configuration being corrupted at some byte.
> 
> Please do this. With v1.2.2, select from the _File_ menu _Restore Default Config_, then flash v1.3 and enable voltage again.

Yes, this works. Now I'm able to use 1.3
Now I've try to activate ESC Smart in combination with GPS. This will result in a dead COMport after Reset button on RP2040Zero.
ESC alone and GPS alone is working fine.
This behaviour is in 1.3 and actual master

ParkerEde, 2025-06-07:
> With commit [9d73597](https://github.com/dgatf/msrc/commit/9d735977b5ef099b9aa5f665dc537a4abb5321fd), I've added GPIO 15 to restore default config if is grounded at boot. This is only available from current (development) version.

nice. if you deliver an uf2 file, I will test. I tried to build it self, but this failed, sorry.

dgatf, 2025-06-07:
> nice. if you deliver an uf2 file, I will test. I tried to build it self, but this failed, sorry.

See development [builds](https://github.com/dgatf/msrc/wiki/01.-Binaries)

dgatf, 2025-06-07:
> Now I've try to activate ESC Smart in combination with GPS. This will result in a dead COMport after Reset button on RP2040Zero.
ESC alone and GPS alone is working fine.
This behaviour is in 1.3 and actual master

Will check that.

ParkerEde, 2025-06-07:
> > nice. if you deliver an uf2 file, I will test. I tried to build it self, but this failed, sorry.
> 
> See development [builds](https://github.com/dgatf/msrc/wiki/01.-Binaries)

well done, it works. GPIO15 to ground and RESET on RP2040Zero and  default settings are loaded.

ParkerEde, 2025-06-09:
Perfect, I will check this evening 👍