dgatf/msrc #168: Spektrum XBUS GPS Not working

Testspektrum, 2025-06-22:
hello.
I wanted to build a GPS for Spektrum, I uploaded the program to the board and configured it, but it doesn't work.
The GPS BN-220 transmits the following data:  $GNGLL, $GNRMC, $GNVTG, $GNGGA, $GNGSA, $GPGSV, $GLGSV
In U-center I see satellites and a strong signal, but the board doesn't work.
I'm posting Debug:
XBUS (222) Address: 16 Packet: 0x16 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
XBUS (222) Address: 17 Packet: 0x17 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
XBUS (222) Address: 16 Packet: 0x16 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
XBUS (222) Address: 17 Packet: 0x17 0x0 0x0 0x0 0x0 0x0 0x0 0x0

dgatf, 2025-06-22:
MSRC is not reading any data from the GPS module. Check circuit connections. Note that to configure your GPS module with the baudrate selected in MSRC you have to reboot the GPS module, then MSRC and then module will be configured. You can try both NMEA and UBLOX protocols.

dgatf, 2025-06-22:
The Tx led (blue) has to be blinking. Otherwise the GPS is not sending any data. The GPS configuration is not working. You can configure it with with ublox. Disconnect GPIO 16. Select the same baudrate, NMEA protocol and it must output GGA and RMC commands.

dgatf, 2025-06-22:
Check that you have connected:

- GPS Tx -> GPIO 6
- GPS Rx -> GPIO 13

Testspektrum, 2025-06-23:
GPS is connected correctly TX to 6 and RX to 13 but still not working

dgatf, 2025-06-25:
I suspect is the baudrate change. Please try this firmware version, which does not set/change the baudrate in the GPS module. Note you have to use MSRC Link v1.4 or development with it.

[MSRC-RP2040_no_baudrate.zip](https://github.com/user-attachments/files/20901847/MSRC-RP2040_no_baudrate.zip)

dgatf, 2025-06-25:
I've reduced the selection of baudrates to 115200, 57600, 38400, 9600, to avoid blocking the GPS Rx. Download updated firmware and MSRC Link.

dgatf, 2025-06-25:
> I will test it on the plane this weekend and let you know if it blocks

The GPS Rx is only used at boot, so it won't get blocked after.

closed 2025-06-25 by dgatf

Testspektrum, 2025-06-30:
Test done, everything works fine 😊