dgatf/msrc #170: GPS: Number of satellites and PDOP

ParkerEde, 2025-06-27:
GPS: Is it also possible to output sensors that show the number of satellites found and the PDOP value? 
Or do these sensors already exist and if so, what are they called?

Many thanks for this great project.

Best regards

dgatf, 2025-06-27:
Which radio system are you using?

Sats are available in xbus, srxl2, jeti, elrs and smartport.

PDOP is available with UBLOX protocol. Though not populated to any receiver protocol.

It can be added to smartport and jeti.

ParkerEde, 2025-06-27:
oh sorry. I use S.PORT
S.PORT/Smartport and Jeti would be great.

dgatf, 2025-06-27:
Ok. I'll add it.

ParkerEde, 2025-06-29:
nice, well done.
A question about the Sats sensor 5103. I know that if the value is greater than 100, then the 3D Satfix is present and therefore you have to subtract 100 from the sensor value supplied. With 13 satellites, 113 would then be delivered from the sensor value. With 5103 I get the value e.g. 1300, so do I have to divide the value by 100?
Is there generally a list of all delivered sensor values with units and explanations?

dgatf, 2025-06-29:
The value of the satellites sensor is the number of satellites detected by tje GPS module. Fix value should not affect the number of sats (no need to substract 100). This sensor has no decimals, so precision must be 0. (default). For PDOP, precision must be set to 0.00, to divide it by 100 as it has two decimals. Note you can change the name of the sensors to Sats, Fix, PDOP...

ParkerEde, 2025-06-29:
> The value of the satellites sensor is the number of satellites detected by tje GPS module. Fix value should not affect the number of sats (no need to substract 100). Sats value should be from 3 to maybe 20. This sensor has no decimals, so precision must be 0. (default). For PDOP, precision must be set to 0.00, to divide it by 100 as it has two decimals. Note you can change the name of the sensors to Sats, Fix, PDOP...

With my M8N module, I always get the number of satellites x 100 with the 5103 sensor, e.g. 500 for 5 or 1300 for 13 satellites.

PDOP is fine.

dgatf, 2025-06-29:
Please check if in MSRC Link log it appears the number of sats multiplied by 100. Just to be sure, check precision in sats sensors is set to 0.-

dgatf, 2025-06-29:
Stupid bug. Fixed with commit 29cd799. Download updated firmware.

closed 2025-06-29 by dgatf