dgatf/msrc #174: Smart ESC does not recognise a signal

ParkerEde, 2025-07-17:
I have tried to use a Smart ESC via a Frsky as well as via Jeti. After switching on, the ESC beeps cyclically and signals that it is not receiving a signal. I used the firmware 1.4.4 and the corresponding configuration tool. Enclosed are the logs.

[Frsky-S.Port-SmartESC.log](https://github.com/user-attachments/files/21302456/Frsky-S.Port-SmartESC.log)

[Jeti-SmartESC.log](https://github.com/user-attachments/files/21302458/Jeti-SmartESC.log)

I followed the wiring exactly.
There's just one thing I'm not quite sure about. Do PIN 12 and PIN13 have to be bridged to the signal connection of the throttle channel of the receiver? Or is it sufficient to use only PIN12, as I don't use thrust reverser?

dgatf, 2025-07-17:
I'll check it. No, you don't need to connect anything to gpio 13 if reverse is not used.

ParkerEde, 2025-07-18:
When I recorded the LOG, I also moved the throttle stick. Thr is always set to 0, so that should be the error.
I have also connected the Smart ESC directly to the receiver throttle channel for testing, then the controller switches freely. So that can't be the problem. I then connected the GAS SignalPIN from the RX to GPIO12 and GPIO 4&5 to the signal line to the SmartESC.

dgatf, 2025-07-18:
Yes, I can reproduce. I'll let you know when fixed.

Have you tried  version 1.3 and higher?

ParkerEde, 2025-07-18:
I've tried only1.4.4

dgatf, 2025-07-26:
There was some issues with the new capture edge library. Fixed with commit e91f244.

Please try latest [binaries](https://github.com/dgatf/msrc/wiki/01.-Binaries) (firmware & MSRC Link)

ParkerEde, 2025-07-27:
Thank you very much. I will check this as soon as possible and report back.

dgatf, 2025-07-31:
I assume the issue is solved. Otherwise reopen this issue and provide feedback.

closed 2025-07-31 by dgatf

ParkerEde, 2025-08-01:
Yes, now it works.
However, I still have problems with the display of the current, when 1A is measured, 2.82A is displayed in the radio and also in the LOG. Can the value still be calibrated?

dgatf, 2025-08-02:
That's an issue of the SmartESC current sensor. The current is delivered with 2 decimals precision. MSRC  can't adjust that mismatch. However that's a low current value and current sensors are less accurate at low current levels. Measure the current over 10A and check the mismatch at that level. I expect the mismatch be much lower in percent at higher current levels.

With opentx/edgetx you can set an offset and a ratio on the telemetry sensors.